danielfett/inetbox.py #11: Truma D6e

maseb24, 2022-10-06:
Hallo,

ich habe deine Software nochmal auf einer neuen Raspi Installation aufgesetzt. 

Das Skript starte ich von Hand, den service habe ich erst gar nicht enabled.
Wenn ich nun truma_service als normaler user starte kommen Fehlermeldungen 

pi@raspberrypi:~ $ truma_service
2022-10-06 13:52:32,220  truma.main  INFO 	started
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/serial0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/bin/truma_service", line 8, in <module>
    sys.exit(truma_service.run())
  File "/home/pi/.local/lib/python3.9/site-packages/inetbox/truma_service.py", line 56, in run
    miqro.run(TrumaService)
  File "/home/pi/.local/lib/python3.9/site-packages/miqro/__init__.py", line 578, in run
    service(
  File "/home/pi/.local/lib/python3.9/site-packages/inetbox/truma_service.py", line 21, in __init__
    self.serial = Serial(serial_device, 9600, timeout=0.03)
  File "/home/pi/.local/lib/python3.9/site-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/home/pi/.local/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/serial0: [Errno 13] Permission denied: '/dev/serial0'

Kannst du mir einen Tip geben wie ich weiterkommen könnte.

LG Rudi

danielfett, 2022-10-06:
Hallo Rudi,

da fehlt die Berechtigung, den Port als normaler Benutzer zu nutzen. Du hast zwei Optionen: 

 1. Du kannst die Berechtigung vergeben: `sudo adduser pi dialout` -> **wichtig:** danach ausloggen und wieder einloggen (oder neustarten), sonst ist die Berechtigung nicht aktiv.
 2. Du kannst das Skript auch mit sudo installieren und starten. (Alle Befehle wie in der Readme, aber mit sudo davor.)

danielfett, 2022-10-06:
Hallo Rudi,

read-logfile.py wird nicht installiert, sondern muss direkt gestartet werden. Es ist aber nur für spezielle Anwendungen notwendig.

Was heißt denn genau, dass keine Verbindung stattfindet?

Bist du dir sicher, dass du den richtigen PIN am RJ12-Stecker angezapft hast und dass RX/TX richtig verkabelt sind?

maseb24, 2022-10-06:
Hallo Rudi! Wie meinst Du das mit "findet nicht statt"? Fehlt bei Dir auch diese Zeile? (kopiert aus Daniels README))
2022-10-02 14:21:27,234 inet-lin INFO Received status data from cp plus

Bei mir fehlt sie, und ich bin noch nicht dahinter gekommen, woran es liegt (siehe https://github.com/danielfett/inetbox.py/issues/8).

Ja genau, diese Zeile kommt nicht.

Ich werde als nächstes versuchen die Verbindung zur Truma mit dem RJ12 Splitter zu machen. Dafür fehlt mir aber noch ein weiteres RJ12 Kabel. Muss ich mir erst besorgen. Die direkte Verbindung über die zweite RJ12 Buchse an der Truma scheint nicht zu funktionieren.

Gruß Rudi

danielfett, 2022-10-06:
@7wells Das ist möglich, aber aufgrund der Fehlerbeschreibung eher nicht wahrscheinlich - mal schauen.

Zur Sicherheit: Eure CP Plus hat jeweils das "inet Ready"-Symbol?

danielfett, 2022-10-06:
Ich habe hier einen Hinweis hinzugefügt: 

https://github.com/danielfett/inetbox.py#hardware-requirements

maseb24, 2022-10-08:
ich habe auch CP Plus mit iNetReady im Bus installiert.

ich bekomme die Initialisierung mit dem LIN Bus der Truma nicht zum laufen.

Daniel hat das schon mal jemand hin bekommen, ausserhalb deiner Installation?.
Ich muss passen.

Gruß Rudi

danielfett, 2022-10-08:
Ja, das wurde schon reproduziert. Insofern ist es zumindest kein grundlegender Fehler, sondern irgendwas anders...

Welche Pi-Version benutzt ihr jeweils?

Könnt ihr bitte einmal folgendes ausprobieren?

 * `sudo apt install --no-install-recommends screen` um das Programm screen zu installieren
 * `sudo systemctl disable miqro_truma` um den Dienst zu deaktivieren, falls er schon installiert wurde
 * Pi neustarten (wichtig!)
 * `sudo screen /dev/serial0 9600` starten
 * Dort sollte spätestens alle 10 Sekunden Datenmüll vorbeikommen. Wenn ihr nichts seht, gibt es ein Problem mit der Verbindung. Ihr könnt dann noch /dev/serial1 statt serial0 testen.
 * Mit `Strg+A`, dann `K` kommt ihr wieder aus screen raus.

Kommt Datenmüll rein oder nicht?

danielfett, 2022-10-08:
Und falls keine Daten kommen:
```
sudo apt install wiringpi
sudo gpio -g mode 15 up
```
- dann bitte nochmal probieren.

appi1, 2022-10-09:
hallo
bin am selben Punkt angelangt, bekomme die Initialisierung auch nicht zum laufen.
mit sudo screen /dev/serial0 9600 kommt im Sekundentakt Datenmüll (unleserlich).
Habe Bulleye installiert und wiringpi gibt es nicht mehr als Paket.

maseb24, 2022-10-09:
Hallo,
ich bekomme auch mit sudo screen /dev/serial0 9600 Datenmüll. Auf serial1 bleibt der screen leer. Wiringpi lässt sich ebenfalls nicht installieren. Ich habe auch Bullseye installiert

Mein Raspi ist ein Pi 3 B V 1.2 aus 2015.

Immerhin mal ein Erfolgsergebnis. Aber eine Initialisierung mit dem LIN Bus der Truma ist nicht möglich.

LG Rudi

skrebber, 2022-10-09:
Ich ergänze auch mal, hatte mit Daniel direkt geschrieben:

Truma 6 Gasheizung mit CPplus. Rpi4 mit exakt dem LIN Transceiver aus der readme. Gleiches Fehlerbild wie bei euch: stecke ich den pi an, geht das display auf W255H, ein PR RESET dauert ewig und anschließend ist die Heizung im Display "weg".
Die Zeile "inet-lin INFO Received status data from cp plus" kommt nicht.
Der Datenmüll bei mir sieht aus wie im Anhang.

VG, Sönke
[image]

morawekj, 2022-10-09:
Wenn das CP Plus den Fehler 255h meldet, kann es nicht mehr mit der Combi reden, das heisst es liegt ein Anschlussfehler am LIN vor. Wichtig, das Interface braucht auch die Vbat (12V vom Wohnmobil) Spannung. bitte auch noch mal zusätzlich die Ground am RJ12 Verbinden.

mc0110, 2022-10-10:
Hallo Sönke,

das mit dem GND aus dem Wechselrichter ist sicher kein Batterie-GND. 

Könntest Du mal einen Link zu dem ESP32-Code teilen. Das wäre toll. Ich würde es gerne auf der Plattform zum Laufen bringen.

Danke und LG
Magnus

mc0110, 2022-10-10:
Was meinst Du mit "dem Code aus der WOMOLin Gruppe"

morawekj, 2022-10-10:
Exakt der Fehler, ohne GND passieren komische Dinge. Die TX/RX Pins sind nicht getrennt, es könnten Ausgleichströme fließen, die den Uart im PI zerstören. Die GND verbindung ist deshalb auch wichtig.

skrebber, 2022-10-10:
Es gibt eine Telegram-Gruppe, dort gibt es einen Arduino-Code, der ein CPplus emuliert und über MQTT Status bereitstellt und sich über MQTT steuern lässt.

mc0110, 2022-10-10:
Oh je, das klingt interessant, aber da komme ich wohl nicht an den Code. Aber danke für Deine schnelle Antwort.

skrebber, 2022-10-10:
Direkt auf der Startseite ist der Link zur Telegram-Gruppe. ;-) https://womolin.de
Dort unter Dateien in der Gruppe "Code_v2.ino"

mc0110, 2022-10-10:
Super hilfreich, ich danke Dir!

skrebber, 2022-10-10:
GND vom Truma RJ12 Connector PIN5 auf GND LIN Transceiver UND GND LIN Transceiver auf GND vom pi. Damit alle auf dem gleichen Niveau sind.

skrebber, 2022-10-10:
Der markierte ist LIN pin3. Rechts pin1. Links ist pin6. 

Ist es ohne Bild verständlich? ;-)

skrebber, 2022-10-10:
So jetzt konnte ich testen.
Wenn ich die RX/TX so anstecke, dass der oben gezeigte Datenmüll kommt und ich dann noch GND vom raspi mit mit GND vom LIN Transceiver und Truma Bus verbinde, dann hört der Datenmüll sofort auf. Trenne ich die GND Verbindung, läuft der Datenmüll wieder los.

Wenn ich RX/TX so anstecke, wie ich meine, dass es richtig wäre und zusätzlich die GND Verbindung habe, dann kommt leider gar nix auf der Schnittstelle. Aber auch vertauscht und zusätzlich GND kommt auch nix auf der Schnittstelle.

Ich würde sagen der UART vom pi ist gehimmelt. :-( Oder siehst du das anders, @morawekj?

Jetzt bin ich mit testen erst mal raus, bis ich nächste Woche wieder zu Hause bin und den nächsten pi nehmen kann. Es sei denn, der UART ist doch nicht hin und es hat noch jemand eine weitere Idee, was ich probieren könnte.

morawekj, 2022-10-10:
Verbinde RX und TX, mache eine serielle konsola auf, kein Echo, irgendeine Baudrate z.B. 19200 8N1 software handschake und tippe etwas ein, wenn der Uart OK ist, kannst du deinen Text dann im Terminal lesen.

morawekj, 2022-10-10:
Evtl. ist auch der LIN Transceiver defekt, irgendein Oszi etc. vorhanden ?

skrebber, 2022-10-10:
> Verbinde RX und TX, mache eine serielle konsola auf, kein Echo, irgendeine Baudrate z.B. 19200 8N1 software handschake und tippe etwas ein, wenn der Uart OK ist, kannst du deinen Text dann im Terminal lesen.

Danke Jan, an diesen einfachen Test hatte ich gar nicht gedacht! Aber der UART geht noch. Gut! Aber irgendwie auch nicht. ;-) Dann bin ich nämlich auch noch nicht weiter mit dem debugging...
Der LIN Transceiver ist aber auch ok, wenn ich die RX/TX Kabel auf meinen ESP32 stecke, kann ich alles steuern und Temperaturen lesen. Dann wird der wohl auch ok sein.

morawekj, 2022-10-10:
Habe mal ein universelles LIN Interface das per USB oder UART angesprochen werden kann, zusammen geklickt.
Voll ESD protected und galvanisch getrennt.
https://gitlab.womolin.de/public-repository/ti-ci-lin-interface

mc0110, 2022-10-13:
Gratulation - sieht gut aus! Und Du bist der Erste, der Vollzug meldet. Das motiviert.

skrebber, 2022-10-13:
Gratulation. Kannst du eingrenzen, welcher Schritt/Änderung/etc zum Erfolg geführt hat?

danielfett, 2022-10-13:
Hast du Init ausgeführt am CP Plus? Wenn der control_status nicht von selbst auftaucht, hat das CP Plus noch nicht die Anwesenheit der "iNet-Box" erkannt.

**Aber:** Das kann jetzt auch wirklich ein Versionsproblem sein! Kannst du das ganze mal mit den drei genannten DEBUG-Optionen starten (auf der Kommandozeile, nicht als Service) und berichten, was da kommt? 

Bitte dafür ein neues Issue aufmachen, ist ein anderes Thema.

danielfett, 2022-10-13:
Die Daten unter display_status kommen ca. alle 10 Sekunden, control_status minütlich oder schneller, wenn die Heizung aktiv ist.

danielfett, 2022-10-13:
Das wäre ein neues Thema, bitte. Möglicherweise anderes Protokoll, daher brauche ich die Debug-Logs.

morawekj, 2022-10-13:
Ich habe mal das LIN Interface fertig gemacht und spontan 20 Stück bestellt. Ob das mit dem USB Port und Timing funktioniert, muss man testen, das sind aber kein 5 $ Unterschied. lässt sich auch mal als Stand Alone USB Uart verwenden.
Der CP2102N kann eigentlich 3 MBs, sollte eigentlich für die 9K6 reichen, aber der Teufel liegt im Detail.

[image]

danielfett, 2022-10-17:
Du kannst nur Werte aus control_status setzen. Wo vent_mode in den inetbox-Steuerpaketen kodiert ist (also Lüftung unabhängig von der Heizung), habe ich noch nicht rausgefunden.

danielfett, 2022-10-17:
Nein, das scheint nicht zu gehen.

skrebber, 2022-10-18:
Ich bin wieder zurück und konnte weiter testen und das Problem lösen:
mein Rpi4 ist das Problem gewesen. Ich habe nur den Rpi4 gegen einen RpiZero W getauscht und es ging (fast) sofort. Interessanterweise funktioniert der Kurzschluss von 14&15 beim Rpi4 trotzdem. Da kann ich die gesendeten Zeichen direkt zurück sehen. Aber am LIN UART kommt gar nichts.
Also an alle, die keine Daten bekommen: ich würde den pi tauschen.

Vielleicht auch für den einen oder anderen interessant: nach pi-Tausch kam sofort der Datenmüll im screen an. Jedoch hat Daniels truma_service keinen sync bekommen (DEBUG_LIN=1 anschalten, dann sieht man's).
Alle Kabel abgezogen und ansteckt, truma_service neugestartet, alles nichts gebracht.
Wie es dann ging: alles angesteckt und so den pi neugestartet, danach hatte er sofort sync. Ob es am Neustart während die Schnittstelle schon angeschlossen war oder es Zufall war, weiß ich nicht.

Vielen Dank an Daniel für die tolle Software!

skrebber, 2022-10-18:
Ich habe den alten pi Zero W rumliegen gehabt und verwendet.

maseb24, 2022-10-20:
ich habs wohl auch hinbekommen, dass mein pi als 3. Gerät von der Truma akzeptiert wird.

nach starten des Skripts kommt es zu einer Verbindung zur Truma

> 
2022-10-20 12:15:55,177  truma.main  WARNING 	Service configuration for truma not found in 'services' section of configuration file miqro.yml. Using empty configuration.
2022-10-20 12:15:55,178  truma.main  INFO 	started
2022-10-20 12:15:55,180  truma.main  INFO 	Using serial device /dev/serial0
2022-10-20 12:15:55,182  truma.main  INFO 	Loop stats:
2022-10-20 12:15:55,183  truma.main  INFO 	 - Loop(_update_online_status) called 0 times, average duration 0.0s, load=0%
2022-10-20 12:15:55,184  truma.main  INFO 	 - Loop(send_status) called 0 times, average duration 0.0s, load=0%
2022-10-20 12:15:55,254  truma.main  INFO 	MQTT connected, client=<paho.mqtt.client.Client object at 0x75ecc478>, userdata=None, rc=0
2022-10-20 12:15:55,255  truma.main  INFO 	Subscribing to ...
2022-10-20 12:15:55,255  truma.main  INFO 	  - service/truma/set/#
2022-10-20 12:15:55,256  truma.main  INFO 	  - service/truma/enabled
2022-10-20 12:16:02,407  inet-lin  INFO 	Received status data from cp plus
2022-10-20 12:16:02,409  app  WARNING 	Unknown status buffer type b'\n\x15'
2022-10-20 12:17:09,939  inet-lin  INFO 	Received status data from cp plus
2022-10-20 12:17:09,940  app  WARNING 	Unknown status buffer type b'\n\x15'
2022-10-20 12:18:01,803  inet-lin  INFO 	Received status data from cp plus
2022-10-20 12:18:01,804  app  WARNING 	Unknown status buffer type b'\n\x15'
2022-10-20 12:18:55,196  truma.main  INFO 	Loop stats:
2022-10-20 12:18:55,197  truma.main  INFO 	 - Loop(_update_online_status) called 1 times, average duration 0.00216s, load=0%
2022-10-20 12:18:55,198  truma.main  INFO 	 - Loop(send_status) called 60 times, average duration 6.423333333333333e-05s, load=0%
2022-10-20 12:19:08,148  inet-lin  INFO 	Received status data from cp plus
> 

Im MQTT Explorer kommt nur der topic service/truma/online =1

Was könnte ich noch probieren um im MQTT Explorer die anderen topics zu empfangen

LG Rudi

danielfett, 2022-10-20:
@maseb24 Das sieht gut aus. Du müsstest auch Daten im MQTT bekommen. Vielleicht nochmal INIT machen, während die Anwendung läuft?

ak68-hub, 2023-10-07:
> Ja, das wurde schon reproduziert. Insofern ist es zumindest kein grundlegender Fehler, sondern irgendwas anders...
>
> Welche Pi-Version benutzt ihr jeweils?
>
> Könnt ihr bitte einmal folgendes [...]

Ich kämpfe weiterhin, nach Ausführung Deiner Anweisungen kommt Datenmüll, 
als scheint die Serial0 Daten zu empfangen.

pi@raspzero2:~ $ sudo screen /dev/serial0 9600
�����`ua�����H�5<ua���������`������y�50`������M���������`������y0����������7v�b`��$�a�������|��� !�-���`ua�b`���`���|��荬5���y�|���������5= �MHM����5�3&7v�b`���
                                                                                2$���`ua�����H�5<ua���������`������y�50`����������4|��M������`������y���`ua�������4|���������5= �MHM�v̵
                      2$a��0��4|=���u���5��y��|��� �&b��

... aber ich kann jetzt den truma_service nicht mehr ausführen:
OSError: [Errno 16] Device or resource busy: '/dev/serial0'
Offensichtlich läuft das Programm screen noch

Wie kann ich dieses wieder schließen ? 
Ich komme mit Strg + A nicht heraus, auch mit Strg + C  nicht
Ein Reboot hat Nichts gebracht !
sudo screen stop funktioniert auch nicht

ak68-hub, 2023-10-08:
Service läuft wieder......
jedoch bei "DEBUG_LIN=1 truma_service" weiterhin keine Synchronisation möglich (Init noch einmal durchgeführt !)
2023-10-08 15:14:23,660  inet.lin  DEBUG        in < 00 ac not a proper sync -wait for sync-
2023-10-08 15:14:23,663  inet.lin  DEBUG        in < 30 60 not a proper sync -wait for sync-
2023-10-08 15:14:23,666  inet.lin  DEBUG        in < ff ff not a proper sync -wait for sync-
2023-10-08 15:14:23,670  inet.lin  DEBUG        in < ff ff not a proper sync -wait for sync-
2023-10-08 15:14:23,712  inet.lin  DEBUG        in < 00 ac not a proper sync -wait for sync-
2023-10-08 15:14:23,715  inet.lin  DEBUG        in < 34 7c not a proper sync -wait for sync-
2023-10-08 15:14:23,718  inet.lin  DEBUG        in < ff ff not a proper sync -wait for sync-
2023-10-08 15:14:23,722  inet.lin  DEBUG        in < ff ff not a proper sync -wait for sync-
2023-10-08 15:14:23,764  inet.lin  DEBUG        in < 00 ac not a proper sync -wait for sync-
2023-10-08 15:14:23,767  inet.lin  DEBUG        in < 3d 20 not a proper sync -wait for sync-
2023-10-08 15:14:23,770  inet.lin  DEBUG        in < 35 21 not a proper sync -wait for sync-
2023-10-08 15:14:23,774  inet.lin  DEBUG        in < a1 8d not a proper sync -wait for sync-
2023-10-08 15:14:23,816  inet.lin  DEBUG        in < 00 ac not a proper sync -wait for sync-
2023-10-08 15:14:23,820  inet.lin  DEBUG        in < ee 33 not a proper sync -wait for sync-
2023-10-08 15:14:23,823  inet.lin  DEBUG        in < 37 76 not a proper sync -wait for sync-
2023-10-08 15:14:23,826  inet.lin  DEBUG        in < 1a 62 not a proper sync -wait for sync-
2023-10-08 15:14:23,868  inet.lin  DEBUG        in < 00 ac not a proper sync -wait for sync-

MQTT zeigt weiterhin cp_plus_status=waiting :(
Ich habe keine Ahnung, wo das Problem liegen könnte
Da Datenmüll bei Daniel´s Anleitung ankommt, scheint ja die serielle Verbindung korrekt zu sein !

Um weitere Ideen wäre ich sehr dankbar :)

ak68-hub, 2023-10-15:
> Ich habe den alten pi Zero W rumliegen gehabt und verwendet.

Wie sind Deine Einstellungen unter /config.exe ?
enable_uart=1 ? wäre mini-uart beim zero 1/2 =0 ?
core_freq=250 ?

Danke
Andreas

ak68-hub, 2023-10-15:
ERFOLG: durch folgende Maßnahmen habe ich ERSTMALS eine Verbindung zur Truma D6E:

[image]

Zum Erfolg führte:
In /boot/config.txt die Zeilen ergänzt:
core_freq=250
enable_uart=1
dtoverlay=pi3-miniuart-bt
und dann noch: sudo systemctl disable hciuart