danielfett/inetbox.py #23: Hardware Question | ESP32

nightcat91, 2022-10-29:
Hi,

would it be possible to get this working on an ESP 8266 or ESP 32, maybe also using ESPHome?

Thanks

mc0110, 2022-11-03:
Hi,

there is an upython port to the ESP32 from my side that I am currently testing and bug-fixing. Hopefully, I will publish this on github in the next few weeks. The connection to a smarthome-center is via MQTT. It has been discussed with Daniel that we don't want to let the projects run apart. Therefore the mqtt-topics and the payloads will be kept the same.

mc0110, 2022-11-04:
@7wells 
I'm using mac-os. My recommendation for the IDE is  [Thonny](https://thonny.org/). There are installation packages for all three platforms. But you don't really need that, I'll send you a ZIP archive with an ESP32-BIN and a short manual (it's still in German, but that shouldn't be a problem, correct?).
[Archiv.zip](https://github.com/danielfett/inetbox.py/files/9939127/Archiv.zip)

mc0110, 2022-11-04:
I'll cross my fingers

mc0110, 2022-11-04:
Yes

mc0110, 2022-11-04:
But you have to use a 4M ESP32

mc0110, 2022-11-04:
V3 is the new one - I haven't tested it with this version

mc0110, 2022-11-04:
@7wells 
I haven't asked Daniel now. I will not open my repro until next week. So I'll keep answering you here. It is only one issue. 
My bin file is a copy from my ESP32, so it also contains the Python. You can also flash a Python first (see instructions from microPython.org) and then install the sources individually. I will send you an archive with the sources in a moment. Then you can also go the second way.

mc0110, 2022-11-04:
[Archiv.zip](https://github.com/danielfett/inetbox.py/files/9940338/Archiv.zip)

mc0110, 2022-11-04:
[Archiv.zip](https://github.com/danielfett/inetbox.py/files/9940494/Archiv.zip)

Yes, you are right, good hint. truma_serv is the major file, will be started with the include in main.

mc0110, 2022-11-04:
You should see the dictionary at the left side. There you can tag them all together - pls. follow the new Helpfile

mc0110, 2022-11-04:
seeing the light at the end of the tunnel

mc0110, 2022-11-04:
The second "Kurzanleitung" is explaining the installation with Thonny also

mc0110, 2022-11-04:
all

mc0110, 2022-11-04:
lib directory and 7 files in the root-directory
lib, boot.py, main.py, conversions.py, inetboxapp.py, lin.py, tools.py and truma_serv.py

mc0110, 2022-11-04:
Ok - first I understand, there is no connection to the CPplus at the moment! Correct?
There was a connection to your MQTT-server running. You see the "HOMEASSISTANT-AUTO DISCOVERY", but this is not really important for your setup. Then you see normal communication to the MQTT, which stopped because a weak wifi-connection.

mc0110, 2022-11-04:
Hi,

dann können wir es auch in Deutsch machen. 
@nightcat91 - alles was Du sagst verstehe ich total gut. Ich habe selbst 4 ESP32 im Auto auf ESPHOME laufen und mega gute Erfahrungen auch bei mir zuhause damit gemacht. Aber.... zunächst war die Herausforderung, ein wirklich komplexes Protokoll umzusetzen. Da bot sich uPython an. 
Ich empfehle, das nicht auf einem ESP32 mitlaufen zu lassen, denn wenn die Puffer (6 Frames) übertragen werden, ist der ESP kurzzeitig wirklich ausgelastet. Daher habe ich auch meine Pläne, es nochmals auf ESPHOME zu portieren, komplett verworfen, insbesondere da es (zumindest bis jetzt) wirklich gut läuft. Daher würde ich hier den eigenständigen ESP empfehlen - und der Stromverbrauch steht in keinem Verhältnis zu einem RPI.

mc0110, 2022-11-04:
Hi,

ich habe mir das gerade nochmals in ESPHOME angeschaut. Das wesentliche Problem ist, dass es in C++ (in dem ESPHOME-environment) umgesetzt werden muss. Da ist das Projekt "als Einstieg" nicht wirklich geeignet. Ich bin ein ESPHOME-Fan, aber leider waren alle Testprojekte für die 'custom component' bei mir nicht lauffähig. Keine gute Vorraussetzung für ein solches Projekt und die API wird vom Entwicklerteam selbst als "unstable" deklariert. Daher habe ich mich für eine "Insellösung" entschieden. Mir war es wichtig, das der ESP32 viel weniger Strom braucht als ein RPI und das die Lösung keine SD-Karte benötigt. Bei dem Projekt wird nix auf die Karte geschrieben, sollte entsprechend stabil sein. Der MQTT-Server ist sowieso an Bord, warum also nicht so realisieren. Probiert es aus, ich bin damit "jetzt" mehr als zufrieden, wenn ich auch manche Bedenken nachvollziehen kann.

Schönes Wochenende
Magnus

mc0110, 2022-11-05:
Yes please

nightcat91, 2022-11-10:
@7wells 
Thanks for asking about my status.
I was not able to fully set everything up yet, cause my LIN Transceiver board was delayed.
But I was following the communication between you and mc0110 and I will try to get the software working once I get the board. If I encounter any problems I will talk to you in your repo.

For this issue here I will close this since it was answered and solved, thanks for all the work from everyone involved!!

closed 2022-11-10 by nightcat91