danielfett/inetbox.py #24: display_status missing

trinler007, 2022-11-02:
Hallo,

vielen Dank für die tolle Software und die damit verbundene Arbeit!

Leider habe ich ein kleines Problem. Meine Umgebung:

inetbox.py Version 0.3
Truma Combi D6E
CP Plus Bedienteil (iNet ready)
RPi Zero mit dem genannten LIN to UART Transceiver 
Als MQTT-Broker nutze ich den bereit vorhandenen Cerbo GX. 

Der Transceiver ist eingeschleift zwischen CP Plus und Truma. Verhält sich aber identisch, wenn ich ihn direkt am zweiten Port der Heizung anschließe.

Ich erhalten keine "display_status" Meldungen. "control_status" und die anderen Topics sind vorhanden. Publishing und damit verbunden die Steuerung funktioniert ebenfalls.

Irgendeine Idee?

Danke!

Markus

mc0110, 2022-11-03:
Hallo Markus,

ich habe bei meiner CPplus das selbe Phänomen. Sie sendet keine 0x20-0x22 Frames (das sind die Statusmeldungen). Dementsprechend habe ich auch keinen display-status. Ich vermute, es liegt an der Version der Softwarestände der CPplus. Meine Versionsnummer der CPplus ist C4.03.00 - 0050.00. Aber es ist nur eine Vermutung.

trinler007, 2022-11-03:
Das ist schade, ich hatte gehofft mehr Informationen und häufigere Updates zu bekommen. Bei nächster Gelegenheit schau ich nach der Version meiner CPplus.

trinler007, 2022-11-08:
Bei mir ist der Stand:
C4.03.00
H6.00.02
T23.70.0
Truma Combi D6 mit E und ohne display_status

Gibt es noch mehr Erfahrungen zu Unterschieden bei den FW/HW-Kombinationen?