danielfett/inetbox.py #28: ALDE use case [validation + might be useful for others]

stepir, 2022-12-12:
Hey

I have an Alde system with a 3010 control panel (3010 413 but who also has the color version 3010 613 is in the same scenario) - this is a closed system

[image]

but it looks like I can upgrade *just* the control panel to a 3020 HE which has a TIN bus interface:

[image]

If that is true I'm planning to use inetbox.py to control my Alde system remotely.

Before trying to borrow a 3020HE control panel from my RV dealer, I wanted to check if anyone has suggestion or think this might or might not work? A normal inet implementation would lead to connect the box to the boiler, but it look like from the second image that this could also be accomplished using the control panel interface?

Thoughts?

stepir, 2022-12-13:
this is a closer pic of the 3020 panel 

[image]

wheelbarrow1980, 2023-01-20:
I have the 3020 system in my van and am trying to get this working in my van. Have connected the lin bus to the boiler (not the panel) but from what I have read, you could connect it to either (as per inet instructions). 

However my truma_service is not working and I am getting checksum errors on the pi and "inet panel comms" failure messages on the 3020 control panel. Will open a separate issue for this and link here for tracking...

wheelbarrow1980, 2023-01-20:
Here is the issue link:

#33

stepir, 2023-01-20:
hey @wheelbarrow1980 I very interested in your findings and will look close at the posted issue. Thanks!!

stepir, 2023-03-28:
[image]

it will be you, myself and a protocol analyser for a while....

wheelbarrow1980, 2023-04-19:
How's the decoding going @stepir ? Anything I can do to help?

stepir, 2023-04-22:
I haven’t even started - but first issue is that my ext panel port is connected to my CBE control panel. I’ve tried doubling wires and connected both (alde Smart Control + CBE panel).

while both works individually they will not work well together. If you change one setting from the first, it will not be updated on the second.

look like the protocol/connection is supporting only one point of control

wheelbarrow1980, 2023-05-05:
@stepir - how were you going to do the decoding part? I have a Truma Inet Box added to my Alde 3020 now and I have a pi setup with the truma service installed and the lin transceiver connected. How do capture the serial comms to then debug?

stepir, 2023-05-12:
I'm slowing getting out of other duties and just had some time tonight to give some reading - it looks I'll have to look after this

https://www.civd.de/en/artikel/ci-bus-board-management-system/

for a second I had hope to find available documentation but it's not the case. However based on online information it looks like it's still a LIN bus implementation. So I'm somehow back on the original track of leveraging what I read here 

https://github.com/muccc/WomoLIN/issues/10

stepir, 2023-05-13:
> @stepir - how were you going to do the decoding part? I have a Truma Inet Box added to my Alde 3020 now and I have a pi setup with the truma service installed and the lin transceiver connected. How do capture the serial comms to then debug?

have yo seen the documentation?

Debugging

If you want to enable debugging, you can set the environment variables DEBUG_LIN=1, DEBUG_PROTOCOL=1, and DEBUG_APP=1, to debug the LIN bus (byte level communication), the protocol layer (handing LIN bus specifics), and the application layer (handling the actual data), respectively.

Example:

DEBUG_LIN=1 truma_service

Specify LOG_DIR to additionally write all logs to a file in the specified directory. Make sure that the directory exists and that the user running the service has write access to it.

Important: Enabling all log options will result in a lot of output. Make sure to disable them again after debugging.

You can also specify debug settings in the miqro.yml file:

services:
  truma:
    log_dir: /var/log/truma
    debug_app: True
    debug_lin: True
    debug_protocol: True