danielfett/inetbox.py #29: cp_plus_status always "waiting"

wolfganghuse, 2022-12-14:
Hello, i get data for topic `service/truma/display_status` but `cp_plus_status` is always `waiting`.

I tried RESET at CP Plus, but no change.

Any idea ?

wolfganghuse, 2022-12-16:
I tried to work on timings on my serial connection (provided by port on FT4232H) but still there is no registration happening. I am using native USB connection with pyftdi, also tried different latency-settings. I have plenty of possibilities to configure the connection, maybe we find a suitable way to get this running...

Maybe my CP Plus has a different version which does not work with registration process ?
Version-Strings:
C04.03.00
0050.00
H1.02.01
0340.0b

I see a lot of not-synced blocks but then again always usable data.
Within the Log-File there are two blocks of readable data, after that i run a PR SET on the CP Plus... maybe someone can help on this ?

[inet.log](https://github.com/danielfett/inetbox.py/files/10246232/inet.log)

skrebber, 2022-12-16:
do you have a GND-connection between all components?

wolfganghuse, 2022-12-16:
12V/GND Stepdown -> Raspi -> USB -> FTDI Breakout -> RX/TX/GND -> TJA1020 -> 12V & LIN/GND (RJ12) -> Splitter

So i guess, as best as possible...  Splitter is used from Daniels Link in the Readme

wolfganghuse, 2022-12-17:
Hi... i changed to MPSSE Mode on the serial connection, and now i don´t receive any `not a proper sync -wait for sync-` anymore... Also registration works directly and i can control the heater as expected.
It´s just some minor change in truma_service.py and lin.py

`
from pylibftdi import Device                                                        
                                                                                              
BITMODE_MPSSE = 0x02                                                                          
INTERFACE_D = 4                                                                               
                                                                                              
DIRECTION_MASK = 0x01  
                                  
self.serial = Device(interface_select=INTERFACE_D)                                    
self.serial.baudrate = baudrate                                                       
self.serial.ftdi_fn.ftdi_set_bitmode(DIRECTION_MASK, BITMODE_MPSSE)   `

Within lin.py i needed to replace the .flush and empty-buffer calls...

ImmutablLimited, 2024-08-15:
>It´s just some minor change in truma_service.py and lin.py

I know this was a long time ago... is there any chance you could share the updated trauma_service.py and lin.py having a very similar problem to you.

DerMatze3279, 2025-03-31:
@wolfganghuse I am running into a similar issue. Any chance to get your updated files trauma_service.py and lin.py? Thanks in advance.