danielfett/inetbox.py #33: Alde 3020 - Checksum error

wheelbarrow1980, 2023-01-20:
Hi, firstly I wanted to say this is awesome work and thank you!

I have installed the scripts and connected my pi to my Alde 3020 (connected to the boiler, not behind the panel.) 

When running the truma_service I get two main errors:

`inet.app  ERROR        Status buffer does not start with preamble, expected b'\x00\x1e\x00\x00"\xff\xff\xffT\x01', got b'\x00*\x00\x00"\xff\xff\xffT\x01'`

AND then:

`2023-01-20 13:06:49,745  inet.protocol  INFO    Responding to 08 message (updates_to_send={})!
2023-01-20 13:06:49,866  inet.lin  WARNING      → → checksum error: Checksum error (received 3c, calculated aa)
2023-01-20 13:06:49,926  inet.lin  WARNING      → → checksum error: Checksum error (received 55, calculated 00)
2023-01-20 13:06:52,522  inet.protocol  INFO    Received read by identifier request.
2023-01-20 13:06:52,812  inet.protocol  INFO    Responding to 08 message (updates_to_send={})!
2023-01-20 13:06:52,932  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa)
2023-01-20 13:06:53,173  inet.protocol  INFO    Responding to 08 message (updates_to_send={})!
2023-01-20 13:06:54,636  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa)
2023-01-20 13:06:54,877  inet.protocol  INFO    Responding to 08 message (updates_to_send={})!
2023-01-20 13:06:55,177  inet.protocol  INFO    Responding to 08 message (updates_to_send={})!
2023-01-20 13:06:55,297  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa)
2023-01-20 13:07:05,569  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa)
2023-01-20 13:07:06,721  inet.lin  WARNING      → → checksum error: Checksum error (received 3c, calculated aa)
2023-01-20 13:07:06,782  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa)
2023-01-20 13:07:16,652  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa)
2023-01-20 13:07:16,983  inet.lin  WARNING      → → checksum error: Checksum error (received 3c, calculated aa)
2023-01-20 13:07:17,043  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa`

I also see an error on the control panel:

"iNet Connecton Fail"

Does the 3020 use slightly different serial comms to the 3010 or did I miss something?

Happy to help with testing and debugging comms if needed...

wheelbarrow1980, 2023-01-20:
LIN DEBUGS:

> 2023-01-20 13:19:26,324  truma.main  WARNING    Service configuration for truma not found in 'services' section of configuration file miqro.yml. Using empty configuration.
> 2023-01-20 13:19:26,324  truma.main  INFO       started
> 2023-01-20 13:19:26,325  truma.main  INFO       Opening serial device /dev/serial0 in exclusive mode
> 2023-01-20 13:19:26,331  truma.main  INFO       Loop stats:
> 2023-01-20 13:19:26,331  truma.main  INFO        - Loop(_update_online_status) called 0 times, average duration 0.0s, load=0%
> 2023-01-20 13:19:26,333  truma.main  INFO        - Loop(send_status) called 0 times, average duration 0.0s, load=0%
> 2023-01-20 13:19:26,333  truma.main  INFO        - Loop(commit_updates) called 0 times, average duration 0.0s, load=0%
> 2023-01-20 13:19:26,337  truma.main  INFO        - Loop(send_update_status) called 0 times, average duration 0.0s, load=0%
> 2023-01-20 13:19:26,337  truma.main  INFO        - Loop(send_cp_plus_status) called 0 times, average duration 0.0s, load=0%
> 2023-01-20 13:19:26,347  inet.lin  DEBUG        in < 00 55 3c 7f 06 b2 00 17 46 30 03 37 → processing
> 2023-01-20 13:19:26,347  inet.lin  DEBUG        → → identified as transportlayer MASTER → SLAVE
> 2023-01-20 13:19:26,347  inet.lin  DEBUG        TRANSPORTLAYER FRAME master → slave
> 2023-01-20 13:19:26,347  inet.lin  DEBUG           node address: broadcast
> 2023-01-20 13:19:26,347  inet.lin  DEBUG           single frame, expected bytes: 5
> 2023-01-20 13:19:26,348  inet.lin  DEBUG           payload: 00 17 46 30 03
> 2023-01-20 13:19:26,348  inet.lin  DEBUG           service id: Read by Identifier
> 2023-01-20 13:19:26,348  inet.lin  DEBUG           → potentially handled by protocol!
> 2023-01-20 13:19:26,351  truma.main  INFO       MQTT connected, client=<paho.mqtt.client.Client object at 0x7f84f198b0>, userdata=None, rc=0
> 2023-01-20 13:19:26,352  truma.main  INFO       Subscribing to ...
> 2023-01-20 13:19:26,352  truma.main  INFO         - service/truma/set/#
> 2023-01-20 13:19:26,352  truma.main  INFO         - service/truma/enabled
> 2023-01-20 13:19:26,376  inet.lin  DEBUG        in < 00 55 7d 01 06 f2 17 46 30 03 00 75 → processing
> 2023-01-20 13:19:26,376  inet.lin  DEBUG        → → identified as transportlayer SLAVE → MASTER
> 2023-01-20 13:19:26,376  inet.lin  DEBUG        TRANSPORTLAYER FRAME slave → master
> 2023-01-20 13:19:26,376  inet.lin  DEBUG           node address: 01
> 2023-01-20 13:19:26,376  inet.lin  DEBUG           single frame, expected bytes: 5
> 2023-01-20 13:19:26,377  inet.lin  DEBUG           payload: 17 46 30 03 00
> 2023-01-20 13:19:26,377  inet.lin  DEBUG           positive response to Read by Identifier
> 2023-01-20 13:19:26,407  inet.lin  DEBUG        in < 00 55 3c 7f 06 b2 00 de 41 01 02 a4 → processing
> 2023-01-20 13:19:26,407  inet.lin  DEBUG        → → identified as transportlayer MASTER → SLAVE
> 2023-01-20 13:19:26,407  inet.lin  DEBUG        TRANSPORTLAYER FRAME master → slave
> 2023-01-20 13:19:26,407  inet.lin  DEBUG           node address: broadcast
> 2023-01-20 13:19:26,407  inet.lin  DEBUG           single frame, expected bytes: 5
> 2023-01-20 13:19:26,408  inet.lin  DEBUG           payload: 00 de 41 01 02
> 2023-01-20 13:19:26,408  inet.lin  DEBUG           service id: Read by Identifier
> 2023-01-20 13:19:26,408  inet.lin  DEBUG           → potentially handled by protocol!
> 2023-01-20 13:19:26,457  inet.lin  DEBUG        in < 00 55 7d 00 55 03 → processing
> 2023-01-20 13:19:26,457  inet.lin  WARNING      → → checksum error: Checksum error (received 03, calculated aa)
> 2023-01-20 13:19:26,460  inet.lin  DEBUG        in < 45 0b not a proper sync -wait for sync-
> 2023-01-20 13:19:26,464  inet.lin  DEBUG        in < ff ff not a proper sync -wait for sync-
> 2023-01-20 13:19:26,467  inet.lin  DEBUG        in < ff ff not a proper sync -wait for sync-
> 2023-01-20 13:19:26,497  inet.lin  DEBUG        in < 00 55 c4 aa 0a ff ff ff ff ff ff 86 → processing
> 2023-01-20 13:19:26,497  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,527  inet.lin  DEBUG        in < 00 55 85 00 08 90 ff ff ff ff ff e1 → processing
> 2023-01-20 13:19:26,527  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,557  inet.lin  DEBUG        in < 00 55 06 00 00 2f f5 ff ff ff ff d4 → processing
> 2023-01-20 13:19:26,558  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,587  inet.lin  DEBUG        in < 00 55 47 44 0b ff ff 2a 00 20 44 da → processing
> 2023-01-20 13:19:26,588  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,616  inet.lin  DEBUG        in < 00 55 d3 fa 0a 03 ff ff ff ff ff 24 → processing
> 2023-01-20 13:19:26,617  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,646  inet.lin  DEBUG        in < 00 55 55 00 00 2f f5 fc 08 ff ff 80 → processing
> 2023-01-20 13:19:26,647  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,676  inet.lin  DEBUG        in < 00 55 d6 ff 0d ff ff d2 0a f5 84 c4 → processing
> 2023-01-20 13:19:26,677  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,707  inet.lin  DEBUG        in < 00 55 5b d2 0a ff 7f ff 7f 00 00 c8 → processing
> 2023-01-20 13:19:26,707  inet.lin  DEBUG        → → not handled by protocol
> 2023-01-20 13:19:26,738  inet.lin  DEBUG        in < 00 55 3c 7f 06 b2 00 17 46 07 0c 57 → processing
[... cut ...]
> 2023-01-20 13:19:28,141  inet.lin  DEBUG        TRANSPORTLAYER FRAME master → slave
> 2023-01-20 13:19:28,141  inet.lin  DEBUG           node address: 03
> 2023-01-20 13:19:28,142  inet.lin  DEBUG           single frame, expected bytes: 5
> 2023-01-20 13:19:28,142  inet.lin  DEBUG           payload: 00 1f 00 00 ff
> 2023-01-20 13:19:28,142  inet.lin  DEBUG           service id: unknown (0xb9)
> 2023-01-20 13:19:28,142  inet.lin  DEBUG           → potentially handled by protocol!
> 2023-01-20 13:19:28,142  inet.protocol  INFO    Received heartbeat request.

danielfett, 2023-01-20:
There is no code for the Alde 3020 yet. It will not work until somebody finds the time to decode the Alde-specific messages.

wheelbarrow1980, 2023-01-20:
I am happy to spend some time on this myself. Would be good to get a primer though if you have something you can share @danielfett ?

stepir, 2023-01-20:
I'm happy to join the wagon - unfortunately my Alde is a 3010 and my only option is to use the external display connector behind the panel. Nothing is documented so I'm just hoping the protocol used is LINBUS as well.

hiwo64, 2023-02-24:
Hi, first off all I wanted to say this is great work and thank you!
I have installed a ADLE 3020 in my RV and so my question, what is the status of the code? Can i help in any way? @wheelbarrow1980

craigpearson, 2023-03-15:
I'm keen on jumping on the band wagon for this too, also have an ALDE 3020. Would anyone have any pointers on how to go about decoding the ALDE 3020? Would be happy to spend some time to adapt this

stepir, 2023-03-22:
I still think the easiest would be getting a used Alde Smart Control device (connect to the external panel connection of the Alde display) and try to reverse engineer the protocol used (most probably CI-BUS)

wheelbarrow1980, 2023-04-25:
ok, so I have managed to get my hands on one of these:
[image]
Have connected and can control the Alde 3020 via the truma app on the truma app via bluetooth. What would be the next steps to get the comms decoded? Can I just useinetbox.py and capture the packets when the pi is connected to a TIN bus port?

hliebscher, 2025-07-13:
i have a complete capture from ALDE 3030 and Truma Aventa

978,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
979,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
980,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
981,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
982,00,55,16,D6,00 00 03 00 8E 02 83 00,,,,,,,,,,
983,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
984,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
985,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
986,00,55,15,55,B4 00 BE 00 E6 00 00 FF,,,,,,,,,,
987,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
988,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
989,00,55,3C,3C,7F 06 B2 00 17 46 07 0C,,,,,,,,,,Settings transfer via LIN transport layer
990,00,55,3D,7D,,,,,,,,,,,
991,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
992,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
993,00,55,15,55,B4 00 BE 00 E6 00 00 FF,,,,,,,,,,
994,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
995,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
996,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
997,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
998,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
999,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
1000,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1001,00,55,3C,3C,7F 06 B2 00 DE 41 01 01,,,,,,,,,,Settings transfer via LIN transport layer
1002,00,55,3D,7D,,,,,,,,,,,
1003,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1004,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1005,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1006,00,55,16,D6,00 00 03 00 8D 02 82 00,,,,,,,,,,
1007,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1008,00,55,3C,3C,7F 06 B2 00 DE 41 02 01,,,,,,,,,,Settings transfer via LIN transport layer
1009,00,55,3D,7D,,,,,,,,,,,
1010,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1011,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1012,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1013,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
1014,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1015,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1016,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1017,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1018,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
1019,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1020,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1021,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1022,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1023,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
1024,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1025,00,55,3C,3C,7F 06 B2 00 17 46 00 0C,,,,,,,,,,Settings transfer via LIN transport layer
1026,00,55,3D,7D,,,,,,,,,,,
1027,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1028,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1029,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1030,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
1031,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1032,00,55,3C,3C,7F 06 B2 00 17 46 00 1F,,,,,,,,,,Settings transfer via LIN transport layer
1033,00,55,3D,7D,03 06 F2 17 46 00 1F 00,,,,,,,,,,
1034,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1035,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1036,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1037,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
1038,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1039,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1040,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1041,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1042,00,55,16,D6,00 00 03 00 8E 02 82 00,,,,,,,,,,
1043,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1044,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1045,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
[... 18 lines cut ...]
1064,00,55,3C,3C,7F 06 B2 00 17 46 04 0C,,,,,,,,,,Settings transfer via LIN transport layer
1065,00,55,3D,7D,,,,,,,,,,,
1066,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1067,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1068,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1069,00,55,16,D6,00 00 03 00 8D 02 83 00,,,,,,,,,,
1070,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1071,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1072,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1073,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1074,00,55,16,D6,00 00 03 00 8C 02 83 00,,,,,,,,,,
1075,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1076,00,55,3C,3C,7F 06 B2 00 DE 41 22 00,,,,,,,,,,Settings transfer via LIN transport layer
1077,00,55,3D,7D,12 06 F2 DE 41 22 00 00,,,,,,,,,,
1078,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1079,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1080,00,55,15,55,BE 00 BE 00 E6 00 00 FF,,,,,,,,,,
1081,00,55,16,D6,00 00 03 00 8C 02 83 00,,,,,,,,,,
1082,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1083,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,
1084,00,55,13,D3,18 00 00 00 00 FF FF FF,,,,,,,,,,
1085,00,55,15,55,BE 00 BE 00 DC 00 00 FF,,,,,,,,,,
1086,00,55,16,D6,00 00 03 00 8C 02 82 00,,,,,,,,,,
1087,00,55,18,D8,FE FF FF FF FF FF FF FF,,,,,,,,,,Settings update requested by CP Plus
1088,00,55,03,03,00 00 00 80 00 00 00 FF,,,,,,,,,,