danielfett/inetbox.py #35: _on_enable takes 2 positional arguments but 3 were given

AndreasL384, 2023-07-30:
Hi,

I installed your software today following the readme file.
When starting the truma_service, I get the following output:

> andreas@raspberrypi:~ $ truma_service
2023-07-30 15:22:59,567  truma.main  INFO       started
2023-07-30 15:22:59,572  truma.main  INFO       Opening serial device /dev/serial0 in exclusive mode
2023-07-30 15:22:59,576  truma.main  INFO       Loop stats:
2023-07-30 15:22:59,577  truma.main  INFO        - Loop(_update_online_status) called 0 times, average duration 0.0s, load=0%
2023-07-30 15:22:59,577  truma.main  INFO        - Loop(send_status) called 0 times, average duration 0.0s, load=0%
2023-07-30 15:22:59,578  truma.main  INFO        - Loop(commit_updates) called 0 times, average duration 0.0s, load=0%
2023-07-30 15:22:59,579  truma.main  INFO        - Loop(send_update_status) called 0 times, average duration 0.0s, load=0%
2023-07-30 15:22:59,580  truma.main  INFO        - Loop(send_cp_plus_status) called 0 times, average duration 0.0s, load=0%
2023-07-30 15:22:59,612  truma.main  INFO       MQTT connected, client=<paho.mqtt.client.Client object at 0x76312520>, userdata=None, rc=0
2023-07-30 15:22:59,613  truma.main  INFO       Subscribing to ...
2023-07-30 15:22:59,617  truma.main  INFO         - service/truma/set/#
2023-07-30 15:22:59,618  truma.main  INFO         - service/truma/enabled
2023-07-30 15:22:59,838  truma.mqtt  ERROR      Caught exception in on_message: _on_enable() takes 2 positional arguments but 3 were given
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/home/andreas/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/home/andreas/.local/lib/python3.9/site-packages/miqro/__init__.py", line 421, in _on_message
    handler(self, payload)
TypeError: _on_enable() takes 2 positional arguments but 3 were given
2023-07-30 15:25:59,580  truma.main  INFO       Loop stats:
2023-07-30 15:25:59,581  truma.main  INFO        - Loop(_update_online_status) called 1 times, average duration 0.004874s, load=0%
2023-07-30 15:25:59,582  truma.main  INFO        - Loop(send_status) called 348 times, average duration 6.132758620689659e-05s, load=0%
2023-07-30 15:25:59,582  truma.main  INFO        - Loop(commit_updates) called 1521 times, average duration 3.68343195266273e-05s, load=0%
2023-07-30 15:25:59,583  truma.main  INFO        - Loop(send_update_status) called 1521 times, average duration 0.0001977172912557534s, load=0%
2023-07-30 15:25:59,583  truma.main  INFO        - Loop(send_cp_plus_status) called 1521 times, average duration 0.00010838724523339853s, load=0%

MQTT states are:
enabled (null)
online 1 [is set to 0 after some time]

I tried to uninstall and install again without success.
For me it seems to be a version conflict or something like that.

Can you suggest something?

Greets
Andreas

pkoevesdi, 2024-01-28:
I have the same problem and I'll take a try to fix this.

pkoevesdi, 2024-01-28:
So, first quick test without understanding, what I do there: If I change the line 421 in `miqro/init.py` (in the path You find in your error message) from 
```
handler(self, payload)
```
to
```
handler(payload)
```
The error message is gone.
Since I'm very early in trying / using this project, I cannot tell, if it works in general.
But I'd consider this error belonging to this project: https://github.com/danielfett/miqro/issues, only judged by the location of the errournous file.

pkoevesdi, 2024-01-28:
Proactively I also remove `self, ` from line 417, because the function call is similar.

AndreasL384, 2024-02-11:
Hi @pkoevesdi,

I will give it a try in a few weeks, when installing it on my new raspi 4B with Venus OS.